tum-vision/dvo_slam #28: Issue compiling dvo_slam on Indigo.

ramanpreet9, 2015-06-24:
[ rosmake ] Last 40 lineso_slam: 16.0 sec ]                                                                                                                                        [ 1 Active 74/75 Complete ]
{-------------------------------------------------------------------------------
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h:36:24: warning: ‘dvo::core::InvalidDepth’ defined but not used [-Wunused-variable]
   static const DepthType InvalidDepth = std::numeric_limits<DepthType>::quiet_NaN();
                          ^
  make[3]: *** [CMakeFiles/dvo_slam.dir/src/camera_keyframe_tracking.cpp.o] Error 1
  In file included from /home/raman/rosbuild_ws1/package_dir/dvo_slam/g2o/include/g2o/core/block_solver.h:189:0,
                   from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/src/keyframe_graph.cpp:50:
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/g2o/include/g2o/core/block_solver.hpp: In instantiation of ‘bool g2o::BlockSolver<Traits>::buildStructure(bool) [with Traits = g2o::BlockSolverTraits<6, 3>]’:
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/src/keyframe_graph.cpp:942:1:   required from here
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/g2o/include/g2o/core/block_solver.hpp:198:5: error: request for member ‘resize’ in ‘schurMatrixLookup->g2o::SparseBlockMatrixHashMap<MatrixType>::blockCols<Eigen::Matrix<double, 6, 6> >()’, which is of non-class type ‘int’
       schurMatrixLookup->blockCols().resize(_Hschur->blockCols().size());
       ^
  In file included from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/rgbd_image.h:29:0,
                   from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/include/dvo_slam/local_map.h:26,
                   from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/include/dvo_slam/keyframe_graph.h:25,
                   from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/src/keyframe_graph.cpp:21:
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h: At global scope:
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h:33:28: warning: ‘dvo::core::Invalid’ defined but not used [-Wunused-variable]
   static const IntensityType Invalid = std::numeric_limits<IntensityType>::quiet_NaN();
                              ^
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h:36:24: warning: ‘dvo::core::InvalidDepth’ defined but not used [-Wunused-variable]
   static const DepthType InvalidDepth = std::numeric_limits<DepthType>::quiet_NaN();
                          ^
  make[3]: *** [CMakeFiles/dvo_slam.dir/src/keyframe_graph.cpp.o] Error 1
  In file included from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/rgbd_image.h:29:0,
                   from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/include/dvo_slam/keyframe_tracker.h:24,
                   from /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/src/keyframe_tracker.cpp:28:
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h: At global scope:
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h:33:28: warning: ‘dvo::core::Invalid’ defined but not used [-Wunused-variable]
   static const IntensityType Invalid = std::numeric_limits<IntensityType>::quiet_NaN();
                              ^
  /home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_core/include/dvo/core/datatypes.h:36:24: warning: ‘dvo::core::InvalidDepth’ defined but not used [-Wunused-variable]
   static const DepthType InvalidDepth = std::numeric_limits<DepthType>::quiet_NaN();
                          ^
  make[3]: *** [CMakeFiles/dvo_slam.dir/src/keyframe_tracker.cpp.o] Error 1
  make[3]: Leaving directory `/home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/build'
  make[2]: *** [CMakeFiles/dvo_slam.dir/all] Error 2
  make[2]: Leaving directory `/home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/build'
  make[1]: *** [all] Error 2
  make[1]: Leaving directory `/home/raman/rosbuild_ws1/package_dir/dvo_slam/dvo_slam/build'
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package dvo_slam written to:
[ rosmake ]    /home/raman/.ros/rosmake/rosmake_output-20150624-144514/dvo_slam/build_output.log                                                
[rosmake-2] Finished <<< dvo_slam [FAIL] [ 16.02 seconds ]                                                                                                                                                    
[ rosmake ] Halting due to failure in package dvo_slam. 
[ rosmake ] Waiting for other threads to complete.                                                                                                   
[ rosmake ] Results:                                                                                                                                                                                          
[ rosmake ] Built 75 packages with 1 failures.                                                                                                                                                                
[ rosmake ] Summary output to directory         

I think the main provlem is 
g2o::SparseBlockMatrixHashMap<MatrixType>::blockCols<Eigen::Matrix<double, 6, 6> >()’, which is of non-class type ‘int’
       schurMatrixLookup->blockCols().resize(_Hschur->blockCols().size());

I looked at other discussions here and solution for fuerte was sudo apt-get install ros-indigo-libg2o libqglviewer-qt4-dev 

I am on indigo. I tried the  instructions but it fails on me:

raman@raman-ubuntu:~/rosbuild_ws1/package_dir$ sudo apt-get install ros-indigo-libg2o libqglviewer-qt4-dev 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libqglviewer-qt4-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libqglviewer-dev:i386 libqglviewer-dev

E: Package 'libqglviewer-qt4-dev' has no installation candidate

I have libqglviewer installed.  but it still gives me the error..  Any way i can resolve this?

Shyphone, 2015-07-10:
I meet the same error.I suppose it may has some connection with G2O versions.
Have you solved this problem already?

songuke, 2015-07-10:
The issue is because of linker errors in g2o_viewer app. Currently, we workaround this problem by disabling the apps in g2o. For our use, it does not affect dvo_slam.

Go to dvo_slam/g2o/g2o_src/CMakeLists.txt, line 159, and comment out the line 
SET(G2O_BUILD_APPS ON CACHE BOOL "Build g2o apps")

Rebuild.

Shyphone, 2015-07-15:
I changed that file as above, and then try "rosmake -s g2o",
but the CMakeLists.txt was overwrite by initial version.
How can I do

songuke, 2015-07-20:
In g2o/Makefile:

sophus: download
mkdir -p build && cd build && \
cmake -DCMAKE_INSTALL_PREFIX=`pwd`/..  ../$(GIT_DIR) && \
        make install

You can try deleting the download word and recompile. I guess it overwrites
the g2o_src folder.

ramanpreet9, 2016-06-10:
@songuke its on line 161 for me not 159.. perhaps you also deleted a couple of lines? i am trying ot reinstall dvo again and stuck somewhere here again.

songuke, 2016-06-11:
I don't have the old setup to recheck, but it should be as above: disable the g2o_viewer.cpp to skip the compilation error.

ramanpreet9, 2016-06-11:
where should i disable that? i checked the cmakelists.txt of all 4 dvo_slam/core/benchmark/ros  but i didnt find this cpp file being called anywhere there

ramanpreet9, 2016-06-13:
@Shyphone did  u manage to resolve this issue?