tum-vision/dvo_slam #31: Issue compiling dvo_slam on Fuerte

amiltonwong, 2015-08-12:
Hi all,

I had followed the install steps. But came across the following compile error.
https://goo.gl/h5G6JT 

Could someone give me some suggestions to fix it? 

THX~

PS.
My system is Ubuntu 12.04, ROS Fuerte installed.
I could build dvo package successfully in the same machine, so I think there's something missing while compiling dvo_slam and g2o.

jesusbriales, 2015-08-19:
Hi,

having a look at your compilation errors it seems to me that you have both repositories *dvo* and *dvo_slam* in the same *ros* workspace. Because of this, the packages *dvo_core*, *dvo_ros* and *dvo_benchmark* are duplicated.
You can see that at the time of compiling that the files related to *dvo_core*, *dvo_ros* and *dvo_benchmark* are being taken from the *dvo* repository folder (the explanation for this can be found 
[here](http://wiki.ros.org/catkin_or_rosbuild) at the end of the subsection 2.4).
The most simple solution for this would be remove the *dvo* folder completely of this workspace, then the compilation of *dvo_slam* should work.

Good luck!

amiltonwong, 2015-08-19:
@jesusbriales . Thanks for your suggestion, but it didn't work even I completely remove dvo package.

jesusbriales, 2015-08-20:
The build error is not the same, the problem with the name conflicts is gone but the error now is related to the *g2o* library. This error is discussed in the issues #19 and #28, and you can see there that some of the solutions can be:
1. Disable the apps in *g2o* through *CMake*. You can do it easily from the *cmake-gui* with
    `cmake-gui dvo_slam/g2o/build/`
    setting the *G2O_BUILD_APPS* option to false.
    I had this same problem and solved it like this.
2. Install these dependencies, as suggested in #19 (although I did not try this):
    `sudo apt-get install ros-fuerte-libg2o liblapack-dev libblas-dev freeglut3-dev libqglviewer-qt4-dev libsuitesparse-dev libx11-dev`

PS:
I compiled *dvo_slam* with Ubuntu 12.04 and ROS Fuerte as well as the Indigo branch of [jefftee](https://github.com/jefftee/dvo_slam) with Ubuntu 14.04 and ROS Indigo.

amiltonwong, 2015-08-21:
@jesusbriales . Thank you so much. Finally I built it successfully.
The first option (Disable the apps in g2o through CMake) doesn't apply for my case, I still got the same build error
The second one works for my case. After I install those dependencies as you suggest. I built it successfully.  :+1:

ramanpreet9, 2016-06-11:
@jesusbriales 
i used your 1) suggestion to turn the flag off 

but i am still getting the same error. I am on indigo so the 2nd option doesnt work for me.  Any suggestions as how to overcome this issue?