tum-vision/dvo_slam #35: PCL linker issue when compiling jade-devel branch on indigo

itsatim, 2015-10-15:
Hi, I tried compiling the new jade-devel branch on indigo (Ubuntu 14.04), since the openni and openni2 packages are still missing in jade. However, I get the following error:

Linking CXX executable /home/tim/catkin_ws/devel/lib/dvo_benchmark/benchmark
/home/tim/catkin_ws/devel/lib/libdvo_visualization.so: undefined reference to `pcl::ApproximateVoxelGrid<pcl::PointXYZRGB>::applyFilter(pcl::PointCloud<pcl::PointXYZRGB>&)'
collect2: error: ld returned 1 exit status

Does anybody have an idea what's causing this? Wrong version of PCL perhaps?

aginika, 2015-10-23:
@itsatim I hope you may solve this problem with #36

itsatim, 2015-10-26:
@aginika Awesome, this worked! Now it compiles fine. Thanks a lot :)

closed 2015-10-26 by itsatim