tum-vision/dvo_slam #38: benchmark results are not accordance with the results presented in the dvo_slam paper

somebodyus, 2015-10-28:
Dear author,

I successfully compile the dvo_slam package in ROS fuerte on Ubuntu 12.04.5. I use the default ROS openni calibration parameters.

I benchmark the dataset "rgbd_dataset_freiburg1_xyz" using the following command,
roslaunch dvo_benchmark benchmark.launch keep_alive:=true
When the program seems stopped, I press "Ctrl+C" to terminate it. Then I do evaluation using the following command,
rosrun rgbd_benchmark_tools evaluate_ate.py --plot PLOT_ate  --verbose assoc_opt_traj_final.txt  groundtruth.txt

The results of ATE for fr1/xyz are as follows, 
compared_pose_pairs 790 pairs
absolute_translational_error.rmse 0.016561 m
absolute_translational_error.mean 0.014532 m
absolute_translational_error.median 0.013264 m
absolute_translational_error.std 0.007941 m
absolute_translational_error.min 0.000828 m
absolute_translational_error.max 0.049756 m

The results are not accordance with the results presented in your dvo_slam (IROS2013) paper. The RMSE value for fr1/xyz indicated in the Table.III of your paper is 0.011, but I got 0.016561. 

Could you tell me the reason? 

PS: Could you tell me how to find how many key frames that have been set during the SLAM?

Thank you very much.

HeYijia, 2015-12-02:
I also faced the similar problem when I using the dvo package. Did anyone figure out? 
Many thanks in advance.

amesh90, 2016-04-20:
same here 
but the error is much diferent 
compared_pose_pairs 791 pairs
absolute_translational_error.rmse 0.029669 m
absolute_translational_error.mean 0.026008 m
absolute_translational_error.median 0.023651 m
absolute_translational_error.std 0.014277 m
absolute_translational_error.min 0.003234 m
absolute_translational_error.max 0.082670 m

nifengzhiyix2, 2019-06-06:
same question, and the error is different in some datasets. Can anyone help? Many thanks!
such as:
fr1/room: in paper is 0.053, the output is 0.192

ZhouXiner, 2022-02-19:
Me too. In my test, the fr1/room is 0.2448