tum-vision/dvo_slam #40: ERROR: cannot launch node of type [dvo_benchmark/benchmark]: can't locate node [benchmark] in package [dvo_benchmark]

lim0606, 2016-01-22:
Hi, I'm a newbie on both SLAM with RGB-D sensors and ROS. 

I installed ROS indigo on Ubuntu 14.04, and successfully compiled dvo_slam with its jade_devel branch. 

I got an error from ros when I tried to run the benchmark by following https://github.com/tum-vision/dvo_slam/issues/5. 
`ERROR: cannot launch node of type [dvo_benchmark/benchmark]: can't locate node [benchmark] in package [dvo_benchmark]`

I downloaded rgbd_benchmark_tools and dataset from https://github.com/jesusbriales/rgbd_benchmark_tools and https://vision.in.tum.de/data/datasets/rgbd-dataset/download, respectively. 

I tried to run the program by running `roslaunch dvo_benchmark benchmark.launch keep_alive:=true` on the dataset folder where I also built `assoc.txt`.

The error message is as follows,
```
... logging to /home/jaehyun/.ros/log/1200da5c-c0cd-11e5-8841-7446a0bb7401/roslaunch-jaehyun-ETRI-Workstation-9672.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jaehyun-Workstation:53798/

SUMMARY
========

PARAMETERS
 * /benchmark/camera_file: /home/jaehyun/ros...
 * /benchmark/coarsest_level: 3
 * /benchmark/constraint_min_entropy_ratio_coarse: 0.03
 * /benchmark/constraint_min_entropy_ratio_fine: 0.6
 * /benchmark/constraint_min_eq_sys_constraint_ratio: 0.3
 * /benchmark/constraint_search_radius: 5.0
 * /benchmark/estimate_trajectory: True
 * /benchmark/finest_level: 1
 * /benchmark/graph_opt_dense_graph: False
 * /benchmark/graph_opt_final: False
 * /benchmark/graph_opt_final_dense_graph: True
 * /benchmark/graph_opt_final_iterations: 1000
 * /benchmark/graph_opt_final_outlier_weight_threshold: 0.1
 * /benchmark/graph_opt_final_remove_outliers: True
 * /benchmark/graph_opt_iterations: 50
 * /benchmark/graph_opt_min_distance: 0
 * /benchmark/graph_opt_outlier_weight_threshold: 0.1
 * /benchmark/graph_opt_remove_outliers: True
 * /benchmark/graph_opt_robust: True
 * /benchmark/groundtruth_file: /home/jaehyun/git...
 * /benchmark/influence_function: 2
 * /benchmark/influence_function_param: 5.0
 * /benchmark/keep_alive: True
 * /benchmark/lambda: 0.0
 * /benchmark/max_iterations: 50
 * /benchmark/max_rotational_distance: 0
 * /benchmark/max_translational_distance: 0.2
 * /benchmark/min_depth_deriv: 0.0
 * /benchmark/min_entropy_ratio: 0.6
 * /benchmark/min_eq_sys_constraint_ratio: 0.3
 * /benchmark/min_intensity_deriv: 0.0
 * /benchmark/mu: 0.05
 * /benchmark/precision: 0.0001
 * /benchmark/reconstruction: True
 * /benchmark/render_video: False
 * /benchmark/rgbdpair_file: /home/jaehyun/git...
 * /benchmark/scale_estimator: 2
 * /benchmark/scale_estimator_param: 5.0
 * /benchmark/show_estimate: True
 * /benchmark/show_groundtruth: False
 * /benchmark/trajectory_file: /home/jaehyun/ros...
 * /benchmark/use_initial_estimate: True
 * /benchmark/use_multithreading: True
 * /benchmark/use_weighting: True
 * /benchmark/video_folder: /home/jaehyun/ros...
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  / 
    benchmark (dvo_benchmark/benchmark)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
ERROR: cannot launch node of type [dvo_benchmark/benchmark]: can't locate node [benchmark] in package [dvo_benchmark]
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
```

Is there anyone who can help me to run dvo_slam? 

Thanks

buHtiGnioJ, 2016-03-04:
Have you tried this before starting?

cd ~/catkin_ws/
source devel/setup.bash

yvtheja, 2017-06-01:
@lim0606 , getting the same error as your. Seems like rosmake didn't go well. Did you find any solution?

Thanks :)

anchuanxu, 2019-07-05:
I have encountered the same problem as you. Have you solved your problem?

Ja721kai, 2020-05-03:
Pretty old thread.

Usually @buHtiGnioJ 's proposed solution solves this. Try:
```
echo "source ~/AAMR/catkin_ws/devel/setup.bash" >> ~/.bashrc   # if not already added
source ~/.bashrc
```

This link might also help you to setup your environment:
http://wiki.ros.org/melodic/Installation/Ubuntu

Furthermore I assume that you granted execution rights onto the file that contains your node with `chmod +x my_node` since that might be a rather common issue as well.