tum-vision/dvo_slam #54: Can I use  Kinect V2?

YuQingL, 2016-12-07:
@christiankerl 
I have tried dvo_slam from: https://github.com/songuke/dvo_slam. I compiled the dvo_slam package in ROS Indigo on Ubuntu 14.04 successfully. 
Used TUM RGB-D benchmark, I can see right point clouds and camera positions in rviz.

I Installed Kinect v2 driver according to https://github.com/code-iai/iai_kinect2. I changed the code and then stored the 640*480 resolution of color and depth images, when connected to the hand-hold kinectv2.  Used these pictures, I see wrong phenomenon in rviz after runing dvo_slam. In the rivz window,  the point clouds is bad and very vague.  I don't know why.
Is the picture that I use the hand-hold kinect v2 to create  not compatible with dvo_slam?
When shooting objects,dose it have to follow a certain rule?

Can you help me?
thanks a lot~

wudanhxh, 2016-12-13:
Hi,I run the dvo_slam according to  https://github.com/songuke/dvo_slam in ROS Indigo on Ubuntu 14.04.
However,when I execute:
roslaunch launch/benchmark.launch dataset:=/home/wll/dvo_slam-jade-devel/example/launch/rgbd_dataset_freiburg1_xyz
the following errors occurred:
[benchmark-2] process has died [pid 28029, exit code -11, cmd /home/wll/dvo_slam-jade-devel/devel/lib/dvo_benchmark/benchmark __name:=benchmark __log:=/home/wll/.ros/log/aa1a8182-c12e-11e6-8b0b-74e54347b077/benchmark-2.log].
log file: /home/wll/.ros/log/aa1a8182-c12e-11e6-8b0b-74e54347b077/benchmark-2*.log
Could you help me out?Thanks！

songuke, 2016-12-13:
It is unclear what causes this error. Please upload the content of the log
file.

wudanhxh, 2016-12-13:
@songuke 
Thank you for your reply.

I did not find the benchmark-2*.log in the folder.Only the following four log files:
master.log                           rosout-1-stdout.log
roslaunch-wll-Lenovo-G480-27987.log  rosout.log

The detail of the roslaunch-wll-Lenovo-G480-27987.log is as follows:
[roslaunch][INFO] 2016-12-13 22:32:01,765: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2016-12-13 22:32:01,777: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2016-12-13 22:32:01,777: roslaunch starting with args ['/opt/ros/indigo/bin/roslaunch', 'launch/benchmark.launch', 'dataset:=/home/wll/data/rgbd_dataset_freiburg1_xyz/']
[roslaunch][INFO] 2016-12-13 22:32:01,777: roslaunch env is {'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'ROS_DISTRO': 'indigo', 'ROS_LOG_FILENAME': '/home/wll/.ros/log/e947a01c-c140-11e6-8473-74e54347b077/roslaunch-wll-Lenovo-G480-30470.log', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/wll', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'UPSTART_EVENTS': 'started starting', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'PKG_CONFIG_PATH': '/home/wll/dvo_slam-jade-devel/devel/lib/pkgconfig:/opt/ros/indigo/lib/pkgconfig', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/wll/dvo_slam-jade-devel/devel/share/common-lisp', 'CPATH': '/home/wll/dvo_slam-jade-devel/devel/include:/opt/ros/indigo/include', 'LOGNAME': 'wll', 'USER': 'wll', 'PATH': '/opt/ros/indigo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games', 'XDG_VTNR': '7', 'GNOME_KEYRING_CONTROL': '/run/user/1000/keyring-Asu94n', 'CMAKE_PREFIX_PATH': '/home/wll/dvo_slam-jade-devel/devel:/opt/ros/indigo', 'LD_LIBRARY_PATH': '/home/wll/dvo_slam-jade-devel/devel/lib:/opt/ros/indigo/lib', 'LANG': 'zh_CN.UTF-8', 'TERM': 'xterm', 'SHELL': '/bin/bash', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'XAUTHORITY': '/home/wll/.Xauthority', 'LANGUAGE': 'zh_CN:zh', 'COMPIZ_CONFIG_PROFILE': 'ubuntu', 'SHLVL': '1', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'JOB': 'gnome-session', 'TEXTDOMAIN': 'im-config', 'UPSTART_INSTANCE': '', 'CLUTTER_IM_MODULE': 'xim', 'WINDOWID': '71303180', 'SESSIONTYPE': 'gnome-session', 'XMODIFIERS': '@im=fcitx', 'ROS_ETC_DIR': '/opt/ros/indigo/etc/ros', 'GPG_AGENT_INFO': '/run/user/1000/keyring-Asu94n/gpg:0:1', 'HOME': '/home/wll', 'QT4_IM_MODULE': 'fcitx', 'SELINUX_INIT': 'YES', 'XDG_RUNTIME_DIR': '/run/user/1000', 'INSTANCE': 'Unity', 'PYTHONPATH': '/home/wll/dvo_slam-jade-devel/devel/lib/python2.7/dist-packages:/opt/ros/indigo/lib/python2.7/dist-packages', 'SSH_AUTH_SOCK': '/run/user/1000/keyring-Asu94n/ssh', 'VTE_VERSION': '3409', 'ROS_ROOT': '/opt/ros/indigo/share/ros', 'GDMSESSION': 'ubuntu', 'IM_CONFIG_PHASE': '1', 'UPSTART_JOB': 'unity-settings-daemon', 'TEXTDOMAINDIR': '/usr/share/locale/', 'GNOME_KEYRING_PID': '1488', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'ROS_PACKAGE_PATH': '/home/wll/dvo_slam-jade-devel/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks', 'XDG_CURRENT_DESKTOP': 'Unity', 'XDG_SESSION_ID': 'c2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-eb3GWjT98U', '_': '/opt/ros/indigo/bin/roslaunch', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'GTK_IM_MODULE': 'fcitx', 'DESKTOP_SESSION': 'ubuntu', 'UPSTART_SESSION': 'unix:abstract=/com/ubuntu/upstart-session/1000/1491', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg', 'GTK_MODULES': 'overlay-scrollbar:unity-gtk-module', 'OLDPWD': '/home/wll/dvo_slam-jade-devel', 'GDM_LANG': 'zh_CN', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/', 'PWD': '/home/wll/dvo_slam-jade-devel/example', 'QT_IM_MODULE': 'xim', 'ROS_MASTER_URI': 'http://localhost:11311', 'COLORTERM': 'gnome-terminal', 'DISPLAY': ':0', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:', 'XDG_SEAT': 'seat0'}
[roslaunch][INFO] 2016-12-13 22:32:01,778: starting in server mode
[roslaunch.parent][INFO] 2016-12-13 22:32:01,778: starting roslaunch parent run
[roslaunch][INFO] 2016-12-13 22:32:01,778: loading roscore config file /opt/ros/indigo/etc/ros/roscore.xml
[... cut ...]
[roslaunch.pmon][INFO] 2016-12-13 22:32:06,980: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon -1231467712)>
[roslaunch][INFO] 2016-12-13 22:32:06,980: ... shutting down processing monitor complete
[roslaunch][INFO] 2016-12-13 22:32:06,981: done
[roslaunch.pmon][INFO] 2016-12-13 22:32:06,981: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon -1231467712)>
[rospy.core][INFO] 2016-12-13 22:32:06,981: signal_shutdown [atexit]

As a beginner,I'm sorry to trouble you.Thanks!

songuke, 2016-12-13:
I'm not sure what happens after seeing this log.

But there's something you can try
- Check the other three log files. Any useful log message there?
- Check trajectory.txt in example/output. Is it empty or are there some
poses estimated?
- Check your input data.

wudanhxh, 2016-12-14:
1.The  trajectory.txt is empty.
The input data : rgbd_dataset_freiburg1_xyz.I have created  an assoc.txt file with the associate.py script.
2.The detail of the other log files:
 **rosout.log** 
1481639522.619735127  Node Startup

**rosout-1-stdout.log:**
logging to /home/wll/.ros/log/e947a01c-c140-11e6-8473-74e54347b077/rosout.log
re-publishing aggregated messages to /rosout_agg
subscribed to /rosout

**master.log**
[rosmaster.main][INFO] 2016-12-13 22:32:02,082: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2016-12-13 22:32:02,083: Starting ROS Master Node
[xmlrpc][INFO] 2016-12-13 22:32:02,085: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2016-12-13 22:32:02,085: Started XML-RPC server [http://wll-Lenovo-G480:11311/]
[xmlrpc][INFO] 2016-12-13 22:32:02,085: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2016-12-13 22:32:02,085: Master initialized: port[11311], uri[http://wll-Lenovo-G480:11311/]
[rosmaster.master][INFO] 2016-12-13 22:32:02,149: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,151: +PARAM [/roslaunch/uris/host_wll_lenovo_g480__35278] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/constraint_search_radius] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/constraint_min_entropy_ratio_coarse] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/graph_opt_dense_graph] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/finest_level] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/graph_opt_outlier_weight_threshold] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/show_groundtruth] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,223: +PARAM [/benchmark/min_eq_sys_constraint_ratio] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/reconstruction] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/max_translational_distance] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/min_intensity_deriv] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/use_weighting] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/graph_opt_min_distance] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/graph_opt_robust] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/min_depth_deriv] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/use_multithreading] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,224: +PARAM [/benchmark/show_estimate] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/scale_estimator_param] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/graph_opt_iterations] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/constraint_min_eq_sys_constraint_ratio] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/influence_function_param] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/estimate_trajectory] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/graph_opt_final_dense_graph] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,225: +PARAM [/benchmark/camera_file] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/min_entropy_ratio] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/render_video] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/max_iterations] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/graph_opt_final_iterations] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/graph_opt_final_outlier_weight_threshold] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/groundtruth_file] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/video_folder] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/keep_alive] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,226: +PARAM [/benchmark/use_initial_estimate] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/graph_opt_remove_outliers] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/lambda] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/mu] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/graph_opt_final_remove_outliers] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/max_rotational_distance] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/coarsest_level] by /roslaunch
[rosmaster.master][INFO] 2016-12-13 22:32:02,227: +PARAM [/benchmark/graph_opt_final] by /roslaunch
[... cut ...]
[rosmaster.master][INFO] 2016-12-13 22:32:06,641: -SUB [/rosout] /rosout http://wll-Lenovo-G480:49821/
[rosmaster.master][INFO] 2016-12-13 22:32:06,642: -SERVICE [/rosout/get_loggers] /rosout rosrpc://wll-Lenovo-G480:40507
[rosmaster.master][INFO] 2016-12-13 22:32:06,643: -SERVICE [/rosout/set_logger_level] /rosout rosrpc://wll-Lenovo-G480:40507
[rosmaster.main][INFO] 2016-12-13 22:32:06,739: keyboard interrupt, will exit
[rosmaster.main][INFO] 2016-12-13 22:32:06,739: stopping master...
[rospy.core][INFO] 2016-12-13 22:32:06,739: signal_shutdown [atexit]