tum-vision/dvo_slam #55: Problems about running speed

sta105, 2016-12-16:
Dear author,

I have tired to run dvoslam on the sequence captured by myself. I build and run dvoslam by following your instruction.  However, when I observe the mapping process by rviz I find the updating rate is intolerably slow, and the program seems to be extremely lagging behind. Moreover, after processes about a hundred frames, my computer will crash because of running out of memory (I have a nice new laptop). It seems that mapping all the raw point clouds might cause this problem, but I'm not sure. 

Could you tell me how can I fix this running speed problem? Thanks a lot：）

songuke, 2016-12-16:
I had the speed issue before. In my case it is simply due to the
compilation in debug mode. Switching to compile in release mode and it is
fast.

sta105, 2016-12-18:
@songuke 

Thank you for your suggestion, but the running speed is still very slow. Btw, could you tell me how long it will take to process a one minute sequence on your pc?

songuke, 2016-12-18:
I don't remember exactly, but it is at interactive rate on a consumer PC,
i.e., at least a few to 10+ frames per second. This can be roughly
estimated by probing the trajectory.txt file after a few seconds.

Another thing to check is input images. Are they png or jpg or some other
formats (which could be time consuming to read and decode -- however this
is not likely the case as far as I know).

gaobo9109, 2017-01-27:
Hi, I am also having the same issue with speed. I ran the cam_tracker node in dvo_ros package on live kinect stream. I measured the time taken for execution of function _**tracker->match(*reference, *current, transform)**_ in camera_dense_tracking.cpp. Sometimes the execution time can go up to 30s.

zhaozhongch, 2019-11-06:
In my case, if I don't visualize the pointcloud the speed is fast but if I visualize the pointcloud, it will be slow