tum-vision/dvo_slam #60: How to use the package in kinetic

AmirCognitive, 2017-06-06:
Hello,
Thank you for your package though I am still not able to use it.

I followed you installation instruction and downloaded the kinetic package of your code and copied to src folder of catkin, and run the following command, but after that, how to use the package, there is no launch file? and not topic :

amir-ai@amir-comp:~/catkin_ws$ rosmake dvo_core dvo_ros dvo_slam dvo_benchmark
[ rosmake ] rosmake starting...                                                                  
[ rosmake ] Packages requested are: ['dvo_core', 'dvo_ros', 'dvo_slam', 'dvo_benchmark']         
[ rosmake ] Logging to directory /home/amir-ai/.ros/rosmake/rosmake_output-20170606-173808       
[ rosmake ] Expanded args ['dvo_core', 'dvo_ros', 'dvo_slam', 'dvo_benchmark'] to:
['dvo_core', 'dvo_ros', 'dvo_slam', 'dvo_benchmark']
[rosmake-0] Starting >>> catkin [ make ]                                                         
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin     
[rosmake-0] Starting >>> sophus [ make ]                                                         
[rosmake-6] Starting >>> genmsg [ make ]                                                         
[rosmake-5] Starting >>> cpp_common [ make ]                                                     
[rosmake-7] Starting >>> opencv3 [ make ]                                                        
[rosmake-5] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common
[rosmake-4] Starting >>> rosgraph [ make ]                                                       
[rosmake-3] Starting >>> cmake_modules [ make ]                                                  
[rosmake-6] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg     
[rosmake-1] Starting >>> rostime [ make ]                                                        
[rosmake-2] Starting >>> xmlrpcpp [ make ]                                                       
[rosmake-0] Finished <<< sophus ROS_NOBUILD in package sophus
 No Makefile in package sophus     
[rosmake-5] Starting >>> rosclean [ make ]                                                       
[rosmake-6] Starting >>> genlisp [ make ]                                                        
[rosmake-0] Starting >>> genpy [ make ]                                                          
[rosmake-3] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules
[rosmake-3] Starting >>> gencpp [ make ]                                                         
[rosmake-2] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp
[rosmake-2] Starting >>> geneus [ make ]                                                         
[rosmake-5] Finished <<< rosclean ROS_NOBUILD in package rosclean
 No Makefile in package rosclean
[rosmake-4] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph
[rosmake-5] Starting >>> gennodejs [ make ]                                                      
[rosmake-4] Starting >>> rosparam [ make ]                                                       
[rosmake-1] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime  
[rosmake-0] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy        
[rosmake-0] Starting >>> rospack [ make ]                                                        
[rosmake-1] Starting >>> roscpp_traits [ make ]                                                  
[rosmake-6] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp  
[rosmake-6] Starting >>> rosmaster [ make ]                                                      
[rosmake-3] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp     
[rosmake-3] Starting >>> roslang [ make ]                                                        
[rosmake-5] Finished <<< gennodejs ROS_NOBUILD in package gennodejs
 No Makefile in package gennodejs
[rosmake-2] Finished <<< geneus ROS_NOBUILD in package geneus
 No Makefile in package geneus     
[rosmake-5] Starting >>> class_loader [ make ]                                                   
[rosmake-2] Starting >>> message_generation [ make ]                                             
[rosmake-4] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam
[rosmake-4] Starting >>> orocos_kdl [ make ]                                                     
[rosmake-1] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits
[rosmake-1] Starting >>> roscpp_serialization [ make ]                                           
[rosmake-6] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster
[rosmake-6] Starting >>> smclib [ make ]                                                         
[... 159 lines cut ...]
[rosmake-7] Starting >>> tf2_ros [ make ]                                                        
[rosmake-7] Finished <<< tf2_ros ROS_NOBUILD in package tf2_ros
 No Makefile in package tf2_ros  
[rosmake-7] Starting >>> tf [ make ]                                                             
[rosmake-7] Finished <<< tf ROS_NOBUILD in package tf
 No Makefile in package tf                 
[rosmake-7] Starting >>> interactive_markers [ make ]                                            
[rosmake-1] Starting >>> tf_conversions [ make ]                                                 
[rosmake-5] Starting >>> pcl_ros [ make ]                                                        
[rosmake-7] Finished <<< interactive_markers ROS_NOBUILD in package interactive_markers
 No Makefile in package interactive_markers
[rosmake-1] Finished <<< tf_conversions ROS_NOBUILD in package tf_conversions
 No Makefile in package tf_conversions
[rosmake-5] Finished <<< pcl_ros ROS_NOBUILD in package pcl_ros
 No Makefile in package pcl_ros  
[rosmake-5] Starting >>> dvo_ros [ make ]                                                        
[rosmake-5] Finished <<< dvo_ros ROS_NOBUILD in package dvo_ros                                  
[rosmake-5] Starting >>> dvo_slam [ make ]                                                       
[rosmake-5] Finished <<< dvo_slam ROS_NOBUILD in package dvo_slam                                
[rosmake-5] Starting >>> dvo_benchmark [ make ]                                                  
[rosmake-5] Finished <<< dvo_benchmark ROS_NOBUILD in package dvo_benchmark                      
[ rosmake ] Results:                                                                             
[ rosmake ] Built 81 packages with 0 failures.                                                   
[ rosmake ] Summary output to directory                                                          
[ rosmake ] /home/amir-ai/.ros/rosmake/rosmake_output-20170606-173808

zxhou, 2017-06-16:
I have the same problem!
I used catkin_make to compile the source code successfully in Kinetic on Ubuntu 16.04. However, after `rosrun dvo_slam camera_keyframe_tracker`, I got a problem that is 
'couldn't find executable named camera_keyframe_tracker below /home/user/catkin_ws/src/dvo_slam/dvo_slam'.
Could anyone tell me how do you solve the problem. Thanks in advance.

zxhou, 2017-06-27:
Fortunately, I just found the solution to solve the problem!
First, you should open the CMakeLists file in the directory 'dvo_slam' and uncomment the last several line to generate the executable file.
Then, return to catkin_ws directory and catkin_make the program.
Now, you can do `rosrun dvo_slam camera_keyframe_tracker` after connecting to the Kinect camera.

zxhou, 2017-06-27:
@AmirCognitive 
Fortunately, I just found the solution to solve the problem!
First, you should open the CMakeLists file in the directory 'dvo_slam' and uncomment the last several line to generate the executable file.
Then, return to catkin_ws directory and catkin_make the program.
Now, you can do 'rosrun dvo_slam camera_keyframe_tracker' after connecting to the Kinect camera.

zhangduoshou, 2017-07-13:
@zxhou thanks a lot for the nice work